nophead/NopSCADlib #221: Suggested improvement to idler pulley naming convention

martinbudden, 2022-02-23:
GT2 idler pulleys come in a variety of sizes. As well as having different tooth counts and diameters, the also have different widths (to support different belt widths) and different bores. I suggest GT2 idler pulleys are named as follows:

`GT2x<tooth_count>x<width>x<bore>_<toothed|plain>_idler`

so we have, eg:
`GT2x16x7x3_toothed_idler` and `GT2x16x7x3_plain_idler`, for the 16 tooth 7mm wide 3mm bore variants.

Whereas the 25 tooth pulleys for GT2x9 belts with 5mm bore would be `GT2x25x11x5_toothed_idler` and `GT2x25x11x5_plain_idler`.

If you agree, I'll do a PR.

nophead, 2022-02-23:
Seems like it is needed but most of the widths are fractional and somewhat manufacturer depended. Perhaps better to use the belt width it is intended for instead.

martinbudden, 2022-02-24:
Belt width is not really sufficient, since pulleys offer varying clearance for the belts. For example there are 6.5mm and 7mm wide pulleys for 6mm belts, under my system these would be `GT2x16x6p5x3_toothed_idler` and `GT2x16x7x3_toothed_idler`.

nophead, 2022-02-24:
Is it a name or a description? I still haven't recovered from the renaming of the steppers. Perhaps the old names should remain as aliases. Currently GT2 pulleys are named from where I bought them, not the exact width as I don't really care what that is as long as the belt fits and I don't know what it is without looking it up but I do know where I bought them.

martinbudden, 2022-02-24:
Is it a name or a description? - interesting question. I think this does depend a bit on the number of things that are being named. Take our old friends ball bearings - the name encodes data about the the size and type of the bearing. This is necessary because there are a huge number of different types of ball bearings.

The reason I bring this up is that I have been experimenting with different types of pulleys for the MaybeCube. The bearings in 16 toothed pulleys are very small and so not very reliable. So I've been experimenting with 20 tooth pulleys, 25 tooth pulleys, and also looking at a variant for 9mm belts. Once you start having to deal with large numbers of types of pulleys, some sort of naming convention becomes, in my view, essential.

I take the point about the compatibility break. I was not expecting you to merge in the NEMA change so quickly, so all my code is currently broken.

I'm quite happy to make the name change and keep the old names as aliases.

nophead, 2022-02-24:
Yes all my projects are broken and while fixing them I found that OpenSCAD is broken on an old project, so I am going down rabbit holes.

They must be rubbish ball bearings if they wear out.

martinbudden, 2022-02-25:
> They must be rubbish ball bearings if they wear out.

Well, I don't expect manufacturers put the highest quality bearings in GT2 pulleys. But I haven't measured the loads, but I imagine they are being run out of spec on 3D printers anyway - small bearings are typically used at lighter loads and higher rpm than used in 3D printers. (This is also an issue with bicycle bearings - also used at higher loads and lower rpm than in other applications.)

Anyway, from measurements I'm guessing that GT2 16 tooth pulleys use MR63 bearings for the toothed pulley and MR83 bearings for the plain pulley, the spec'ed loads for these are:

MR63
Basic dynamic load rating (C) - 0.208 kN
Basic static load rating (C0) - 0.074 kN

MR83
Basic dynamic load rating (C) - 0.56 kN
Basic static load rating (C0) - 0.179 kN

I'm changing the MaybeCube to use stacked F623 flanged bearings (as on the original HyperCube):
Basic dynamic load rating (C) - 0.63 kN
Basic static load rating (C0) - 0.22 kN

While my 16T toothed pulleys (ie MR63 bearings) are holding out on the BabyCube, I've had one fail on the MaybeCube, but the MaybeCube is working it much harder - the belts are very tight and I've been experimenting with very high accelerations (20,000 and higher) (From comparing Klipper input shaper results, I think the MaybeCube has a more rigid frame than the RatRig, Voron 2.4, SnakeOil and VzBot printers.)

I've borrowed a trick from the Voron 0 and reversed the belts on the MaybeCube, so that the smooth side of the belts is against the flanged bearings (except for two instances on the Y carriages). Note the belt orientation:

[image]

nophead, 2022-02-25:
So, if I understand correctly: The worse case is the MR63, which can handle a belt tension of 74N. It is probably possible to exceed that if you really tighten the belts but all the others seem hard to exceed. Not sure what the optimum belt tension is. I presume the pitch is only spot on at a particular tension. I had an idea to calibrate a printer with a camera and a fine scale by adjusting the belt tension.

martinbudden, 2022-02-25:
I've never thought to put the figures in before now. 

1. Lulzbot recommends a belt tension of 23-27 Newtons ( https://ohai.lulzbot.com/project/taz-6-final-assembly/taz-6/ ), so a static load rating of 74N seems perfectly adequate. 
2. Assume a combined X-carriage, X-rail and Y-carriages mass of 1 kg (way overestimate) and accelerations of 10,000 mm/s<sup>2</sup>, ie 10 m/s<sup>2</sup> and assume all the load goes on one belt, then we have a dynamic load of 10N, again well within spec. This ignores friction, but even allowing generously for friction it seems the bearings are within spec.

So it looks like I had a dodgy bearing on my MaybeCube. Having said that, I'm going to retain my new design with the reversed belts, I think it has several advantages over the previous design. However I will keep the original belt path for the BabyCube.