nophead/NopSCADlib #224: How to make a screw knob for M6 or larger

Lenbok, 2022-03-31:
I need a screw knob for a M6 screw, and thought there may be a suitable component in here (I've never used NopSCADlib before). My attempt to just modify the `tests/screw_knob.scad` demo to take M6 weren't really adequate, as the finger knob radius did not increase. Is there a recommended way to adjust additional parameters such as the knob radius and depth?

nophead, 2022-03-31:
Simply making `knob_r = knob_stem_r + knob_wall` produces this:

[image]

Is that what you need or do you want extra parameters?

Lenbok, 2022-03-31:
That's not too bad, I ended up increasing knob_waves (the ideal size of a knob wave is essentially determined by human finger size), and I also increased the thickness and decreased the stem height for my purpose. I think the last two might be useful to parameterize.

nophead, 2022-03-31:
Most objects in the library are fully parametric. This one is an old piece of code I migrated and not up to the same standard. I will rewrite it to take a parameter list, like the other objects with a function construct the list with sensible default values that can be overridden to taste..

Lenbok, 2022-04-01:
OK, so it was just my bad luck that the first item I try from the library happened to not be parameterized! FYI, here's what I ended up printing (along with the original I'm needing to replicate), in case you wanted to adjust the look and feel at the same time :-):
[image]

closed 2022-04-01 by nophead

Lenbok, 2022-04-01:
Wow, that's great!

nophead, 2022-04-01:
The original use was for some M3 bed levelling screws and then I re-used it for fastening a removable cover with M4 screws, I never anticipated using it to replace broken knobs.