nophead/NopSCADlib #225: [Feature Request] Add USB mini A socket

hadess, 2022-05-30:
Similarly to the `usb_uA` "vitamin", a USB mini A socket would be useful for prototyping cases for existing electronics that use them.

nophead, 2022-05-30:
Yes,  I actually put a usb_uA on the model of the Arduino Nano because I was too lazy to model the mini.

hadess, 2022-05-31:
This *seems* to work. I've only tested the cutout in real life, and it seems to be quite a fair bit bigger than the socket itself, but that seems to match what was done for the mini socket.
```
module usb_miniA(cutout = false) { //! Draw USB mini A connector
    l = 9.4;
    iw1 = 7.8;
    iw2 = 6.07;
    ih1 = 1.05;
    ih2 = 3.10;
    h = 4.00;
    t = 0.4;
    flange_h = 4.55;
    flange_w = 9;

    module D() {
        hull() {
            translate([-iw1 / 2, h - t - ih1])
                square([iw1, ih1]);

            translate([-iw2 / 2, h - t - ih2])
                square([iw2, ih2]);
        }
    }

    if(cutout)
        rotate([90, 0, 90])
            linear_extrude(100)
                offset((flange_h - ih2) / 2 + 2 * panel_clearance)
                    D();
    else
        color("silver") rotate([90, 0, 90]) {
            linear_extrude(l, center = true)
                difference() {
                    offset(t)
                        D();

                    D();
                }

            translate_z(-l / 2)
                linear_extrude(1)
                    offset(t)
                        D();

            translate_z(l / 2 - t)
                linear_extrude(t) difference() {
                    union() {
                        translate([0, h - t - ih1 / 2])
                            square([flange_w, ih1], center = true);

                        translate([0, h / 2 + flange_h / 4])
                            square([iw1, flange_h / 2], center = true);

                        translate([0, h / 2 - flange_h / 4])
                            square([iw2, flange_h / 2], center = true);
                    }
                    D();
                }
        }
}
```

hadess, 2022-05-31:
I used those technical drawings of the micro and mini USB sockets to adapt your original micro USB one.

[image]
[image]

nophead, 2022-05-31:
Thanks, I added it with a few tweaks. I removed the flanges as mini doesn't have them and I fixed the widths as they are inside dimensions, not outside. I also improved the shape.

[image]

nophead, 2022-05-31:
Implemented by cc5654d7d756287f5311e35104003f38a4a94475

closed 2022-06-02 by nophead