nophead/NopSCADlib #232: Build unnecessarily compares tiny (_tn) images

martinbudden, 2022-06-30:
The build process compares the `_tn` images and then deletes the difference files. Although the `_tn` files are small and so quicker to compare, it still takes time and slows down the build.

nophead, 2022-07-02:
The magic compares typically take around 100ms, so I don't see any reason to change this. OpenSCAD takes nearly 2 seconds to do nothing and often takes minutes, so it totally dwarfs the times.

In a build that takes 423 seconds to do the assembly images it only took 2 seconds to do all the small image compares and 6.7 to do the large ones.

nophead, 2022-07-16:
I just pushed a change that logs all the command times in cmd_times.txt.

Please can you confirm the times to run magick are negligible on your system as well.

closed 2024-01-14 by nophead