nophead/NopSCADlib #237: JST XH Connectors left out of BoM

SecareLupus, 2022-09-07:
I ran into an issue today where when used as PCB components, JST_XH connectors get left off the BoM in the assembly instructions.

I've attached an image showing two separate assemblies that include JST_XH on PCBs that otherwise itemise their components.
[image]

Below is the code block for the smaller/simpler of the two assemblies.
```
module DoorBoard_assembly()
assembly("DoorBoard") {
    pcb([
        "DoorBoard",
        "", 
        27.5, 
        32, 
        1.4, 
        0, 
        2.75, 
        5.75, 
        "purple", 
        true, 
        [
            [2,2],
            [25.5,30],
            [2,30],
            [25.5,2]
        ], 
        [
            [19,    12, 90,     "2p54boxhdr",   3,  2],
            [19,    25, 180,    "jst_xh",       2],
            [9,     16, 180,    "gterm35",      6]
        ], 
        [], 
        []
    ]);
}
```

nophead, 2022-09-07:
Yes that is the case for a lot of PCB components as I use them to model ready made PCBs, rather than make my own. The fix is add all the missing vitamin() module calls in such components.

SecareLupus, 2022-09-07:
@nophead, alright, I wanted to submit in case it was an unexpected edge-case. Not a deal-breaker for my use, and if it turns out to be, maybe I'll track down the module calls and submit a pull request. For right now, I don't feel familiar enough with the project to want to get all up in the guts of it. Thanks for the response, feel free to close this ticket or not as project priorities dictate :) Big fan of the library overall!

nophead, 2022-09-07:
I can fix it but not just right now as I am busy with other things.

nophead, 2023-05-08:
Fixed via 06a286dc989e9cc8cf7918049cd16a862b1a9b94

closed 2023-05-08 by nophead