nophead/NopSCADlib #240: SBR16 rails

jeroenrnl, 2022-09-27:
I am working on putting a design of a CNC machine in OpenSCAD and I'm trying to supply this project with anything that I'm missing.

The next part, I'd like to model is SBR16 rails (and possibly SBR12 and SBR20 if I can find detailed enough pictures / measurements online as I do not have any samples myself).

[image]

I'm just doubting where I will put the parts as they fit in a few categories...
### rails ###
This is essentially a steel rod on a mounting rail. I could put it under *rods*, function wise it would better fit in *rails*, however, both the implementation for the rail and the carriage would not fit in the current functions so ```rail(SBR20, 900)``` wouldn't work without significant overhauling of the current implementation. A third option would be creating a new *SBR* category.

### carriage ###
The carriage is really a bearing block, but with a cutout for the rail mount:
[image]
So it could be an addition to *bearing block*, but it could also be added to *rail* or *SBR* if that's the choice above.

I think adding it to *rail* makes the most sense, from an end-user point of view, but it will require some refactoring of the current code. I'd like to hear your take on it before I start ~destructtion~ making changes.

nophead, 2022-09-28:
I think the bearing blocks are similar enough to the existing bearing blocks that can be added by adding a split option. Then I think we need a new class for the rail. I think it is probably too different to shoehorn into the existing rails. The only downside of having a new class is that you can't easily make a machine that can switch from one rail type to another but that is probably an early design decision.

nophead, 2023-07-22:
Added via #258, thanks.

closed 2023-07-22 by nophead