nophead/NopSCADlib #242: Countersunk hole for 3D printed part?

ganzgustav22, 2022-10-11:
I'm trying to make a countersunk hole for 3D printed parts.

I've read the documentation here, but it doesn't seem to have an explanation or example on how to call the function.
https://hydraraptor.blogspot.com/2020/12/sinkholes.html

I've downloaded the NopSCADlib, extracted it, put it into a subdir and included it like this:
`include <NopSCADlib/lib.scad>
`
Now I wanted to make a countersunk hole, like this:
`screw_polysink(type = hs_cs, h = 100, alt = false, sink = 0);`

But I get warnings and an error:

[image]

Could somebody please give a hint of maybe what I might be doing wrong?

nophead, 2022-10-11:
The type parameter should be a screw with a countersunk head, not a screw head type.

ganzgustav22, 2022-10-11:
d'ohhh. Now with "type = M3_cs_cap_screw" it works. Thanks a lot :)

nophead, 2022-10-11:
No problem. Most of my modules and functions that operate on objects have a `type` parameter. It is like `self` in Python or `this` in C++. Perhaps type was not the best name for it but too late to change now.

closed 2022-11-18 by nophead