nophead/NopSCADlib #244: Best way to support internal threads?

mconsidine, 2022-10-17:
Hi,
I am trying to create a threaded tube, with the threads on the inside.  This would essentially be a round nut, but with an arbitrary outside diameter.

The code I am looking at is adapted from one of the examples and is below.  My question is : in preview mode everything compiles without warning.  When I try to render it though, I get a warning that it may not be a valid 2-manifold.

I've probably missed something in the docs or the example - my apologies.  But can someone point me to what would be a "best practice" for accomplishing this?

Thanks in advance,
mconsidine
`
$fn=192;

include <../utils/core/core.scad>

use <../utils/thread.scad>
pitch = 1;
starts = 1;
$show_threads = true;

profile = thread_profile(pitch / 2, pitch * 0.366, 30);

for(female = [false, true]) translate([0, female ? -50 : 0]) 
{
        length = female ? 6  : 10;
        dia = female ? 38.5 : 38.5 - pitch;
        colour = female ? brass : silver;
        
        if (female)
        {
            linear_extrude(6) //trying to create the outer support here
            {
              difference()
              {
                circle(r=(dia/2)+3);  //outer surface
                circle(r=(dia/2)+.1);  //inner surface; +0.1 to catch default "solid" support??
              }
            }
            thread(dia, starts * pitch, length, profile, starts = starts, top = 45, bot = 45, female = female, colour = colour, solid=true, center=false);

        }
        else
        {
            thread(dia, starts * pitch, length, profile, starts = starts, top = 45, bot = 45, female = female, colour = colour, solid=true, center=false);        
        }
}
`

nophead, 2022-10-17:
Not sure why it goes wrong but the errors seem to be at the start and end. Changing to the normal bot = -1, top = -1, seems to generate a manifold result.

I also removed the +.1 from the inner diameter as the thead is designed to fit inside the nominal diameter and has a bit of overlap to avoid coincident surfaces.

mconsidine, 2022-10-17:
Thanks for that feedback/check.  Can you tell me if - aside from any printer miscalibration - an externally threaded plug designed this way should fit into an internally threaded "nut" designed this way?  Ie is there any tolerance that would be expected to be added to one or both pieces to get them to fit?  Or is that already accounted for?
Thanks again ...

nophead, 2022-10-17:
When I make printed parts that screw together I don't use normal thread profiles. I use  a thread with 45 degree slopes and a flat crest. E.g.
```
thread_depth = 0.7;
profile = thread_profile(thread_depth, 0.8, 90);
pitch = 3.5;
```

I make the female thread diameter about 0.2mm bigger than the male to give a bit of clearance and that has always worked well but this is for large screw top containers like this:

[image]

I recently printed an M3 female thread in PLA with no extra clearance and was able to drive a screw into it OK.

mconsidine, 2022-10-17:
Thanks for those insights and that example.  That helps alot.

closed 2022-10-17 by mconsidine