nophead/NopSCADlib #247: [Feature Request] Specify extrusion slot count

mysmartbus, 2022-12-10:
Working on a project that uses 1020 and 2020 t-slot profiles (called "extrusions" in NopSCADlib). The 1020 profiles render correctly for my design but the 2020 does not.

My source for t-slot profiles is 8020.net and they have 2020 profiles that have 2 slots on each side. See [part number 2020-s](https://8020.net/2020-s.html) . The picture below shows the type of 2020 t-slot profile I'd like to be able to use.
[image]

I've been going through the code in vitamins/extrusion.scad and short of doing a complete rewrite, I have not been able to get it to generate the design I'd like to use.

8020.net uses series to make it easy to determine the width of the profile. I.E. 10-series is 1" wide, 15-series is 1-1/2" wide, 20-series is 20mm wide, 25-series is 25mm wide, etc. The height and width specified in vitamins/extrusions.scad would need to evenly divisible by the series value.

Ideas I'm testing:
<dl>
<dt>Adding another argument to module extrusion() in vitamins/extrusion.scad.</dt>
<dd><p>Usage examples:<br>
A) extrusion(E2020, series=10). This would render an E2020 extrusion with 2 slots on each side.<br>
B) extrusion(E2020). This would render an E2020 extrusion with 1 slot on each side.</p>

<p>This method would have the most flexibility as you would not need to duplicate settings in vitamins/extrusions.scad to add another t-slot profile design. But you would have add another item to each of the lists to specify the default series (10-series, 15-series, etc) to use.</p>
</dd>

<dt>Adding an item to the lists in vitamins/extrusions.scad to specify the series to use.</dt>
<dd><p>If the series is set to 10, an E2020 extrusion would render as four E1010 extrusions grouped together in a square measuring 2" x 2". This would give 2 slots per side.</p>

<p>If the series is set to 20, an E2020 extrusion would render as one E2020 extrusion measuring 20mm x 20mm with 1 slot per side.</p>

<p>If the series is set to 15, an E2020 extrusion would not render because 20 cannot be evenly divided by 15.</p>

<p>The problem with this idea is that it could result in a lot of duplicated settings and code.</p>

<pre>
// Current list format
//                  W    H    d1     d2   sq  cw  cwi  t   st  f  recess
E2020  = [ "E2020", 20,  20,  -4.2,  -3,  8,  6,  12,  2,  2,  1, false ];

// With the series item added
//                      W    H    d1     d2   sq  cw  cwi  t   st  f  recess  series
E102020  = [ "E102020", 20,  20,  -4.2,  -3,  8,  6,  12,  2,  2,  1, false,  <b>10</b> ];
E202020  = [ "E202020", 20,  20,  -4.2,  -3,  8,  6,  12,  2,  2,  1, false,  <b>20</b> ];
</pre>
</dd>
</dl>

nophead, 2022-12-10:
Extrusions are all metric outside the US. E2020 is 20mm by 20mm not 2" by 2". Having a series property makes sense but you would also need a duplicate set of definitions with imperial dimensions. I don't think having it as a parameter is the way to go.

nophead, 2022-12-10:
Rather than a series number it might be better to have say E1010 so the code has a way to find the dimensions of the smaller slots, etc.