nophead/NopSCADlib #249: [Feature Request] Please add oscillator crystals

LenStruttmann, 2023-01-28:
Please add oscillator crystals.
Such as: 
https://www.digikey.com/en/products/detail/ecs-inc./ECS-240-20-4DN/281767?utm_adgroup=Crystals&utm_source=google&utm_medium=cpc&utm_campaign=Shopping_Product_Crystals%2C%20Oscillators%2C%20Resonators_NEW&utm_term=&utm_content=Crystals&gclid=Cj0KCQiAic6eBhCoARIsANlox86G0ai6fUx_7o-KqEQVTIYLmhD-CWUmuOvdPNFnlOaLalONcJGrKksaAlB8EALw_wcB

Thanks!

nophead, 2023-01-31:
[image]

[image]

Needs a bit more work to handle cylindrical ones and oscillator modules but I will merge it soon.

LenStruttmann, 2023-01-31:
Awesome!

LenStruttmann, 2023-01-31:
And, if you have time, could you add a SOIC14?  There are two on this HC-SR04 module that I'm modelling.
https://createc3d.com/783-large_default/buy-hc-sr04-ultrasonic-sensor-price-offer.webp

nophead, 2023-02-01:
Implemented by df06d1f57fff64cd6da20684f8371da204f69a82

closed 2023-02-01 by nophead