nophead/NopSCADlib #251: Script error

hotwolf, 2023-02-16:
When I try to run `make_all.py` on my [project](https://github.com/hotwolf/WindowPainter) I get the following Python error:

> magick tmp/WPPulley_assembly.png -trim -resize 1004x1004 -bordercolor #F8F8F8 -border 10 tmp/WPPulley_assembly.png 
> Traceback (most recent call last):
>   File "lib/NopSCADlib/scripts/make_all.py", line 47, in <module>
>     views(target)
>   File "/home/dirk/Projects/WindowPainter/lib/NopSCADlib/scripts/views.py", line 229, in views
>     update_image(tmp_name, png_name)
>   File "/home/dirk/Projects/WindowPainter/lib/NopSCADlib/scripts/tests.py", line 76, in update_image
>     pixels = compare_images(png_name, tmp_name, diff_name)
>   File "/home/dirk/Projects/WindowPainter/lib/NopSCADlib/scripts/tests.py", line 65, in compare_images
>     printf(("magick compare -metric AE -fuzz %d%% %s %s %s" % (fuzz, a, b, c)).split(), output = output)
> NameError: global name 'printf' is not defined

The `magick` call fails, because  ImageMagick (7.1.0-62) does not recognize the color `#F8F8F8` and magick.log is empty:

> $ magick tmp/WPPulley_assembly.png -trim -resize 1004x1004 -bordercolor #F8F8F8 -border 10 tmp/WPPulley_assembly.png 
> magick: missing output filename `-bordercolor' at CLI arg 5 @ error/magick-cli.c/ProcessCommandOptions/755.
> 

So, when I change the `background` variable in `tests.py` to "Gray", the `magick` call succeeds, but the Python error remains:

> openscad -o tmp/WPPulley_assembly.png tmp/png.scad -D$cwd="/home/dirk/Projects/WindowPainter" --viewall --autocenter -D$pose=1 -D$explode=0 --colorscheme=Nature --projection=p --imgsize=4096,4096 -d assemblies/deps/WPPulley.deps --hardwarnings 
> magick tmp/WPPulley_assembly.png -trim -resize 1004x1004 -bordercolor Gray -border 10 tmp/WPPulley_assembly.png 
> magick compare -metric AE -fuzz 5% assemblies/WPPulley_assembled.png tmp/WPPulley_assembly.png assemblies/WPPulley_assembled_diff.png 
> Traceback (most recent call last):
>   File "lib/NopSCADlib/scripts/make_all.py", line 47, in <module>
>     views(target)
>   File "/home/dirk/Projects/WindowPainter/lib/NopSCADlib/scripts/views.py", line 229, in views
>     update_image(tmp_name, png_name)
>   File "/home/dirk/Projects/WindowPainter/lib/NopSCADlib/scripts/tests.py", line 76, in update_image
>     pixels = compare_images(png_name, tmp_name, diff_name)
>   File "/home/dirk/Projects/WindowPainter/lib/NopSCADlib/scripts/tests.py", line 69, in compare_images
>     pixels = int(float(pixels if pixels.isnumeric() else -1))
> AttributeError: 'str' object has no attribute 'isnumeric'

magick.log now contains the following value, but the python code still can't parse the string:

> $ cat magick.log 
> 40560

Do you have any idea why the NopSCADlib scripts don't work in my case?

Thanks for your help,
Dirk

nophead, 2023-02-16:
I have Version: ImageMagick 7.0.8-14 Q16 x64 2018-10-24, perhaps they made incompatible changes. 

According to the latest documentation it shows the border color options still accepts hex but the example has it in quotes and uses lower case hex. Not sure if that is the first problem.

The python error looks like you are using version 2.x.

nophead, 2023-02-16:
Although the usage still says Python 2.7 is supported it looks like it was broken by the introduction of `.isnumeric()` two years ago. Since nobody else has noticed in those two years I think I will change the usage to specify Python 3 only.

Let me know if lower case hex and or adding quotes fixes imagemagick and I will see if that is backwards compatible.

hotwolf, 2023-02-16:
Thanks for the quick help!
I've had the same problem with ImageMagick 7.0.9-26, before I switched to the latest version. Quotes around the hex value work and uppercase hex code is no problem (within quotes).
I'll give Python 3 a try.

hotwolf, 2023-02-16:
Switching from Python 2.7.17 to 3.6.9 resolved all problems. Even the unquoted hex code is not a problem anymore.
So there is no change in the tests.py script needed, if you drop the support for Python 2.7.

nophead, 2023-02-16:
OK I will do that although I don't understand why it makes a difference to needing quotes or not.

nophead, 2023-03-11:
I removed Python 2.7 from the usage. No idea why you had a problem with the command line because `subprocess.call()` should only use the shell if passed `shell = True` and it defaults to `False`. Also it used to work fine on 2.7 for me.

closed 2023-03-11 by nophead