nophead/NopSCADlib #252: Issue using vitamins/extrusion_brackets.scad>

LenStruttmann, 2023-03-01:
I guess that I am confused about how to use NopSCADlib vitamins.

This works fine:
include <NopSCADlib/lib.scad>  
extrusion(E2020, 50);

But this does not:
include <NopSCADlib/lib.scad>  
extrusion_corner_bracket(E2020);

I get the following error:
Compiling design (CSG Tree generation)...
[WARNING: undefined operation (undefined / number) in file screws.scad, line 134](https://github.com/nophead/NopSCADlib/issues/134,C:/Users/busin/Documents/OpenSCAD/libraries/NopSCADlib/vitamins/screws.scad) 
[TRACE: called by 'find_screw' in file extrusion_bracket.scad, line 99](https://github.com/nophead/NopSCADlib/issues/99,C:/Users/busin/Documents/OpenSCAD/libraries/NopSCADlib/vitamins/extrusion_bracket.scad) 
[TRACE: called by 'screw_clearance_radius' in file extrusion_bracket.scad, line 99](https://github.com/nophead/NopSCADlib/issues/99,C:/Users/busin/Documents/OpenSCAD/libraries/NopSCADlib/vitamins/extrusion_bracket.scad) 
[TRACE: called by 'extrusion_corner_bracket' in file AZELD_Assembly-00.scad, line 19](https://github.com/nophead/NopSCADlib/issues/19,D:/OpenSCAD/AZELD/AZELD_Assembly-00.scad) 
Compiling design (CSG Products generation)...

What am I doing wrong?

Thanks

nophead, 2023-03-01:
You have passed an extrusion instead of an extrusion bracket as the type parameter. E.g. `E20_inner_corner_bracket `or `E20_corner_bracket`.

LenStruttmann, 2023-03-01:
Yes, I see that now.  In my newbieness I was confused, not realizing that the parameter "type" means different things for extrusions and extrusion_brackets. Thanks!

As a suggestion, to avoid future newbie questions, you could change the documentation to explicitly specify "extrusion_type" and "extrusion_bracket_type".  Just a thought.

Thanks!  And thanks for the library!  Len

closed 2023-03-01 by LenStruttmann

nophead, 2023-03-01:
I use `type` like `this` in C++ and `self` in Python, so every parametric object in the library has a `type` parameter. The documentation is made from the source code, so it has to match it. If they were given more meaningful names it would make the code a lot more verbose because the type parameter is always heavily used.