nophead/NopSCADlib #253: [ Feature Request] Extrusions/Brackets: Provide support for other screw/nut sizes

LenStruttmann, 2023-03-03:
For extrusions and extrusion brackets, please provide support for other screw and nut sizes.  For example:

https://www.amazon.com/Sutemribor-Hammer-Fastener-Aluminum-Profile/dp/B07FPLZXTF/ref=sr_1_2_sspa?crid=3K4WIBDX2ZQTF&keywords=M4+T+Nuts&qid=1677864030&s=industrial&sprefix=m4+t+nuts%2Cindustrial%2C78&sr=1-2-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExTzRWSTFXTkJQMVExJmVuY3J5cHRlZElkPUEwNzY4MDg1MlM4OFg5NjA5MUhJOSZlbmNyeXB0ZWRBZElkPUEwNDM2OTIxMk1XWldGUTZDVzBDTiZ3aWRnZXROYW1lPXNwX2F0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

I 3D printed several extrusions and found that I could not use them with my M5 nuts.

Thanks!

nophead, 2023-03-03:
Not sure what you mean. The extrusions are supposed to be models of actual extrusions, not something to print. If a nut is sold for 2020 extrusions it should fit in a 2020 extrusion. Does it nearly fit or not at all? If the former perhaps you printer isn't accurate enough. If the  latter perhaps the nuts are wrong for that size of extrusion. 

The extrusions and nuts were contributed by @martinbudden and I haven't checked them myself but I believe them to be accurate. There are different nuts for different extrusions though. It doesn't just depend on the screw size.

LenStruttmann, 2023-03-03:
These nuts:

"20 Series T Sliding Nuts Hammer Nut Block Square Nuts M5 T Nuts for 2020 Aluminum Profiles T Slot 6mm (m5)"
https://www.amazon.com/gp/product/B07V1HX9GS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

...do not fit.

Also, my M5 screws do not fit the bracket created with extrusion_corner_bracket(E20_corner_bracket).

In vitamins/extrusion_brackets.scad, it states:

E20_corner_bracket = [ "E20_corner_bracket", [28, 28, 20], 2, 3, 19.5, **M4_sliding_t_nut**,      E2020t];

This implies, to me, that these extrusions are sized for M4 screws and nuts.

However, since I now know that they are NOT meant to be 3D printed and used, I guess that you don't need to worry about this request.  There are other profiles available on the internet. [Although, it would be nice if the corner brackets were usable without needing to be drilled out.]  I was just being lazy, trying to avoid designing a profile myself.

I'm 3D printing these extrusions because I am very much in the prototyping stage of my project.  The lengths of these extrusions will certainly change and I wanted to not use my aluminum extrusions until I found the final dimensions.

Thanks.

LenStruttmann, 2023-03-03:
I ended up designing my own profile based on the aluminum extrusions that I have on hand.  Took me about 25 minutes.

closed 2023-03-03 by LenStruttmann

nophead, 2023-03-04:
2020 extrusions don't vary much, the nuts and brackets do.  The extrusions with take up to M5 screws I think but it is common to use smaller screws because M5 steel screws are very strong compared to 2020 aluminium extrusions. M4 is normally plenty strong enough.

Looking at the nuts you linked the outside dimensions are all the same, just the screw hole changes so they should all fit the extrusion. In what way do they not fit? What did you change about the extrusions?

I am not at home at the moment but will be next week so I will have a look if I have any M5 nuts for 2020. There are M5 sliding T nuts in the library. How do their dimensions compare with the ones you have?

LenStruttmann, 2023-03-04:
_In what way do they not fit?_

The nuts were too thick, i.e., the extrusion slot did not accommodate the thickness of the sliding nut.

_What did you change about the extrusions?_

I took out an existing aluminum extrusion and my calipers, made some measurements, and built the profile from scratch.

_How do their dimensions compare with the ones you have?_

My sliding nuts have a measured thickness of 4.63 mm.  Without some sort of visual guide, I'm not sure what the measurements sx, ty1, and ty2 refer to and, unfortunately, I don't have time today to dissect the code to see how they are used.