nophead/NopSCADlib #257: Python error: too many values to unpack (expected 2)

hotwolf, 2023-04-27:
I've modified a working NopSCADlib design. The updated design renders fine in OpenSCAD, but when I run `make_all.py`, I get the following error:

```
openscad -D $bom=2 -D $preview=true -o openscad.echo -d bom/bom.deps scad/bom.scad --hardwarnings 
Generating bom ... too many values to unpack (expected 2)
```

My project is located here: https://github.com/hotwolf/DIYLaserBed. The NopSCADlib scripts run fine on "revA", but not on "revB".

Has anyone run into this issue before?  What could be the cause?

Thanks for your help!

nophead, 2023-04-28:
It's a Python error meaning a miss-match between the number of items returned by a function and the number of variables to accept them.  E.g. if you name `a, b = func()` and func returns three values.

I don't understand why there isn't a stack trace to show which line the problem is in.

Do you get any files generated in the bom directory?

hotwolf, 2023-04-28:
I'm not sure anymore, but I think the bom directory was empty. 
In the meantime I've traced through the assembly hierarchy by commenting out lower level assemblies step by step. The problem is in one of my self made vitamins:

```
module flatSection(length=100,center=true) {
    description = str("Flat section: 15mmx2mm, ", length, "mm");
    vitamin(str("flatSection(): ", description));
    
    color("Silver") cube([length,2,15],center=center);    
}
```
    
If I don't instantiate this vitamin, the make_all.py script seems to run through.
But, I still don't understand the cause of the problem.

nophead, 2023-04-28:
It is the colon in the description that is causing the issue.

nophead, 2023-04-28:
The colon is used as a separator between the module call and the description. You have a second one in the description.

hotwolf, 2023-04-28:
I see. Thanks so much for your help!

nophead, 2023-04-28:
I will try to fix the code so you can have colons in the description.

nophead, 2023-04-29:
Fixed by https://github.com/nophead/NopSCADlib/commit/b1592f51d4b16a125d9b28edae56f758fcac2f02

closed 2023-04-29 by nophead