nophead/NopSCADlib #259: Adafruit Feather 405 Module

thestumbler, 2023-07-17:
Here's a quick-and-dirty module I made for the Feather 405 board, in case it comes in handy for anyone.
-Chris

[feather405.scad.txt](https://github.com/nophead/NopSCADlib/files/12074420/feather405.scad.txt)

nophead, 2023-07-22:
Thanks. Added via 523fc9e9a5c8a566bde2bfe8555d5f6dddd8f05b.

I replaced the variables that were only used once by constants for speed. I also added pads for the pin headers and replaced the generic chip with a QFP package.  Lots of other components could be added. I usually add at least the LEDs and the bottons so that holes can be made for them in enclosures.

closed 2023-07-22 by nophead

thestumbler, 2023-07-23:
Thanks for the improvements. That was kind of where I plan to head with this, to make a simple enclosure. I couldn't find any enclosure online that was written in OpenScad, just STL files.