nophead/NopSCADlib #260: [Feature request] Standard PC Motherboards

Slabity, 2023-10-02:
While motherboards look wildly different, the board sizes, standoff locations, IO panel, and PCIe slots are all standardized. I think it would be nice to have some generic models of the common sizes.

I'm trying to find a library that might provide these, but I can't seem to find any at all.

nophead, 2023-10-03:
Feel free to contribute. I don't have any data for motherboards. A full model would require a lot of new components to be modelled. A bare board with screw holes would be fairly easy.

emteeoh, 2023-11-03:
*which* motherboard are you looking for? ATX, Mini-ATX, DTX, Mini-DTX, Mini-ITX, Nano-ITX, pico-ITX, or (I just found out about) Mobile-ITX? Some of those should be easy to find info on. 

Also, I don't think IO locations are well standardised. Modern motherboards come with a stamped steel plate that is about 2" by 6" that has cutouts for all the MB's IO ports and cases have a matching spot for the plate to fit into.  The PCIe slot locations are standardised at least.

emteeoh, 2023-11-03:
https://github.com/nophead/NopSCADlib/pull/261  for Mini-ITX

Slabity, 2023-11-03:
Yea, my bad. I meant to say the I/O shield has a standard size, not the ports themselves. I believe the shield is 6.260" (159mm) wide and 1.760" (44.70mm) tall.

Personally I'm only interested in the popular workstation formats. ATX, Micro-ATX, Mini-ITX, and Extended-ATX are probably the most common ones. I don't think the Nano-ITX or Pico-ITX are used much outside of SoC platforms.

I found a lot of information on the boards on this site: https://www.protocase.com/resources/how-to-design-for-motherboards/

I was able to make a simple ATX board with the standoff locations and I/O shield, but I'm kind of Really Bad™ at working with SCAD and haven't been able to make the PCIe slots. I'm going to keep trying and see if I can make a module that lets you select the format + PCIe slot spacing though. I think I can get something like that to work.

tpimh, 2023-12-28:
[This image from Wikipedia](https://en.wikipedia.org/wiki/File:ATX_ITX_AT_Motherboard_Compatible_Dimensions.svg) can be useful if implementing all common motherboard sizes in a modular manner (only dimensions and hole positions are shown).