nophead/NopSCADlib #265: What is the proper way to add additional pulley, belt, etc. definitions

ganzgustav22, 2024-01-10:
For example I would like to add a pulley for GT2 12mm belts in: https://github.com/nophead/NopSCADlib/blob/master/vitamins/pulleys.scad

I can just do it in that file and it works fine, but how to do it in a cleaner way, i.e. without modyfing the nopscad lib itself?

nophead, 2024-01-10:
You could move your new pulley definition to your own library to avoid modifying NopSCADlib but if you share the definition I could add it to NopSCADlib for you.

ganzgustav22, 2024-01-10:
D'ohh that sounds almost too easy, I'll try. Thanks.

Regarding the definitions: These are some random Aliexpress pulleys, I think they don't follow any standard so I guess it doesn't make sense to add them to the lib.

I have officially Gates approved pulleys ordered though, when they arrive I can measure them and give you the dimensions, these should have standard Gates dimensions.

ganzgustav22, 2024-01-10:
Here are definitions for GT2 belts 12mm and GT3 belts 6,9,12mm:

```
GT2x12 =["GT", 2.0, 12, 1.38, 0.75, 0.254];
GT3x6 = ["GT", 2.0,  6, 1.52, 0.75, 0.254];
GT3x9 = ["GT", 2.0,  9, 1.52, 0.75, 0.254];
GT3x12 =["GT", 2.0, 12, 1.52, 0.75, 0.254];

```

GT3 dimensions:
[image]

nophead, 2024-01-10:
Thanks. I will add them. I will need to change the code because I had assumed GT2 was 2mm pitch but GT3 is also. So the current code would show GT2 on the BOM entry.

nophead, 2024-01-14:
I have added your belts.