nophead/NopSCADlib #287: [Feature Request] add cost calculation to BOM

nerdyjan, 2024-10-06:
You mostly know what a screw cost, or a motor, or a single bearing. Wouldn't it be great to have somehow a way to enter parts prices to get a cost estimation for parts and vitamins in the BOM.
In the first step vitamins prices should be sufficient but as an extension there could be a print cost calculation somehow.

What do you think?

nophead, 2024-10-06:
There is an undocumented method to do this. If you create a `parts.py` file with functions that return the price and an optional URL where to buy it then `bom.py` will create a CSV file which can be imported into a spread sheet. Here is a short example:
```
def nut_M3_nut(qty, nyloc = False):
    if nyloc:
        return 0.0276, 'https://www.orbitalfasteners.co.uk/products/m3-nyloc-nuts-stainless-steel-a2-304-din-985-type-t'
    return 0.0096,  'https://www.orbitalfasteners.co.uk/products/m3-hexagon-full-nuts-stainless-steel-a2-304-din-934'

def screw_M3_cs_cap_screw(qty, length):
    return 0.01, ''

def screw_M3_dome_screw(qty, length):
    return 0.01, ''

def washer_M3_washer(qty):
    return 0.01, ''

def tubing_HSHRNK32(qty, length = 15):
    return 0.01, ''

def iec_IEC_inlet_atx(qty):
    return 1.0, ''

def insert_F1BM3(qty):
    return 0.01, ''

def jack_4mm_shielded(qty, colour_name, thickness, colour = None):
    return 0.1, ''

def mains_socket_Contactum(qty):
    return 3.67, 'https://uk.rs-online.com/web/p/plug-sockets/1752851'
```

nophead, 2024-10-06:
Actually it is documented here: https://github.com/nophead/NopSCADlib/blob/a90eb429338eb5f4f8480393444ef94e156fb3a3/docs/usage.md#costed-boms

nerdyjan, 2024-10-06:
ohh .. working with you library since years .... but missed that, thank you!

closed 2024-10-06 by nerdyjan