nophead/NopSCADlib #288: HT pipes are broken, due to rework of original code ;)

nerdyjan, 2024-10-06:
see attached image, before your rework of the HT pipes the lower end of the t-fitting is a pipe, now it'S just a fitting.
The t fittings are ok, but the pipes missing the "pipe" part ... the fitting is working fine.

[image]

nophead, 2024-10-07:
Addressed by 8cdc04b87ba2ed06918e9f3be858abd7962708c1

nophead, 2024-10-07:
The fitting should really have a test if it is used on its own.

nerdyjan, 2024-10-07:
no, the fitting is never used by it's own. It's always the top (or side) of any Pipe/Junction/etc.

I just checked out your merge and tested it on my project and the fittings in my screenshot are pipes, not just fittings and the error was/is still persistent. i checked out my original code and the screenshots show you, what the problem is.

Your code rework is quite comprehensive and i don't know all the small tips & tricks included in your library. this is why it would cost me to much time to reinvestigate, where the problem occures and why.

can you explain, why my original "not so smart"l code was not appropriate? Your rework is quite comprehensive ;)

DAMAGED:
[image]

OK with original code:
[image]

nerdyjan, 2024-10-07:
and this is the exploded view:
[image]

nerdyjan, 2024-10-07:
and this exploded with the current state of the code (after your rework):
[image]

nophead, 2024-10-07:
I thought I had explained why I changed the code. Basically I don't have
any 3D differences or intersections unless I have to and then they are
wrapped in render. Union is free in OpenCSG and      2D operations very
fast and so is extrude. 3D differences don't scale when there are thousands
of them.

I have no idea why the pipes are missing when they are there in the tests.
I am out for the day so will look when I get home. Please can you post your
code for the broken example.

nerdyjan, 2024-10-07:
I understand ... mainly because of performance ... OK

The pipes are also missing when i run tests.py on your version.

[image]

The code i used for my "case" is:
```
module case_assembly()
    pose(a = [120, 0, 120], t = [0, 0, 0])
    assembly("case") {
        stl_colour(pp2_colour, view_transparency_case) {
            explode(180)
                translate([0, 0, 12])
                    ht_pipe(HT_50_pipe_250);

                translate([0, 0, -100])
                    rotate([0, 0, 180])
                        ht_tpipe(HT_50_40_tpipe);

            explode(-180)
                translate([0, 0, -360])
                    ht_pipe(HT_50_pipe_250);

        }
    }
```

nophead, 2024-10-07:
Your code works for me once I included lib.scad and fixed lib.scad to include ht_pipes.scad instead of  using it.

[image]
I think the bug is an extra unnamed parameter in the call to tube in ht_pipe. That seems to be  silently ignored with the version of OpenSCAD I am using but seems to override the h parameter in the version of OpenSCAd you are using. I will push a fix this evening.

nerdyjan, 2024-10-07:
I'm using OpenSCAD version 2021.01 on linux.
Your guess may hit.

nophead, 2024-10-07:
Hopefully 0989a02d95ed26d137238f3866e8b37c9bf5f2e0 fixes it. I reverted the last change.

closed 2024-10-07 by nerdyjan