nophead/NopSCADlib #292: scs_bearing_blocks preview looks correct, but render is not complete

tpimh, 2024-10-17:
I am using OpenSCAD 2024.10.15.ai20955 (git 335ed8eb2), but this can also be reproduced with earlier versions.

In order to reproduce, add any `scs_bearing_block` to your project and comment out `if ($preview)` line.

For some reason, only the bearing is rendered, but not the other parts of the bearing block.

This is how preview (F5) looks:
[image]

This is how render (F6) looks:
[image]

I expect the two to look identical (like `sbr_bearing_block` does). Not sure why the rendering only works partially.

nophead, 2024-10-17:
None of the vitamins and designed to be rendered with F6. They are only for making assembly views with F5. F6 is used to make STLs for printing and DXfs for routing. In general vitamins are not manifold so CGAL would barf at them and there are no colours with CGAL so it would be pointless. 

I am aware recent versions of OpenSCAD can show colours with manifold but I haven't run a version that new. It would be a lot of work to make the vitamins renderable and I am not sure what is allowed with manifold that isn't with CGAL. CGAL only accepts manifolds and is very strict. Manifold, despite the name, seems less strict at accepting and creating manifolds.

tpimh, 2024-10-17:
Yes, I am currently rendering with Manifold backend (hence the color in F6 render). One of the recent developments in OpenSCAD is the PovRay export: https://github.com/openscad/openscad/pull/5288. The author of the export intends to export the viewport information into the PovRay file, so the result would look similar to OpenSCAD PNG export.

For NopSCADlib it can mean that `povray` can possibly replace `magick`. The process would be "OpenSCAD export POV" -> "PovRay" (instead of "OpenSCAD export PNG" -> "ImageMagick"), would take some much time, but would look nicer. This is all only theoretical at the moment, I've got no code to show, because the only way to get the correct viewport in such export at the moment is patching the exported file. Not impossible, but I would better just wait for the updated export containing the viewport information.

nophead, 2024-10-17:
Yes the problem is OpenSCAD has been around for about 14 years and has only supported colour in the last few months, so this library never anticipated the change. Also the latest versions of OpenSCAD don't support my use of $variables so I have to use the last release to build my objects.

I have no idea how much work it would be to make the vitamins render with F6.

tpimh, 2024-10-18:
Here's a simple spinning animation of Enviro+ that I made for demonstration of PovRay. Looks like nothing is missing.

https://github.com/user-attachments/assets/389da296-b966-46cc-b32b-1c037a24f45d

nophead, 2024-10-18:
The colours aren't correct though. 
[image]

As I don't understand what Manifold's rules are I have no idea what will work and what won't. CGAL doesn't like assembly views because things touch without overlapping, etc. If you want to debug all the vitamins and post a PR then that is OK with me as long as the assembly views with F5 don't change or slow down.

nophead, 2024-10-20:
Also I think we would still need ImageMagick to compare the images to check for differences and show them, etc. It is also used to get the correct size because OpenSCAD's viewall is too small in most cases. I think it scales the bounding sphere to the vertical size of the window. With ImageMagick I crop it to the to the object's pixels and then scale it and add a border. So unless povray can auto zoom getting OpenSCADs viewport isn't sufficient.