nophead/NopSCADlib #293: LED meter font misses the dot

mmuman, 2024-10-19:
After drawing the outline in KiCad, exporting it as SVG and adding stuff manually in OpenSCAD to model a LED meter I only now noticed there was existing models here…

Now, the font used doesn't show the decimal dot as a dot on the digit itself but as an underscore… Could we find another font that has one? [Maybe the one there](https://github.com/darealshinji/7-segment) ? But it's not slanted like the current one.

nophead, 2024-10-20:
I think it may be incompatible licence wise to bundle it but we could have a font parameter to allow any font to be used in a project. I found one that was slanted and had a dp but it was a diamond shape rather than a circle and no obvious licence just a copywrite.

mmuman, 2024-10-20:
Yeah, it's not that urgent anyway, I can live with that on the preview.

tpimh, 2025-05-21:
7 segment fonts are not particularly hard. I can try to implement one in OpenSCAD without using TTF import.

nophead, 2025-05-21:
That would be a good solution.