nophead/NopSCADlib #294: LED meter model differ from measured one

mmuman, 2024-10-19:
when replacing my own model from the measurements I made with the one from the lib, I noticed mine was larger by quite a few mm… i measured 32.7mm total length, 10.5mm width though, so a little smaller. 

Should we account for the larger one? But then the center distance is different as well… :thinking: 

[image]

nophead, 2024-10-20:
It must be a slightly different version because I used them in a project and that is where I got the dimension from. So we would need a different model adding. Not sure what it would be called though!

mmuman, 2024-10-20:
That's the problem with these items, they all clone each other but with varying details… Maybe passing default dimensions and allowing to override them…

nophead, 2024-10-20:
It would be a bit odd to do that because it is a different object, so should have a new object definition as that is how everything else works in the library. Also it makes people aware there are different versions when they use it so they are more likely to not get s surprise when they don't fit.

mmuman, 2024-10-20:
How would we go about naming them? I just home we don't end up with 20 of them.

nophead, 2024-10-20:
You can name with the important dimensions that change. E.g. led_meter23x11x26

mmuman, 2024-10-20:
As I did with the rocker yeah…

nophead, 2024-10-20:
With the rocker you just called it micro and put the dimensions in the description. I was suggesting making the name from the important dimensions as well. Yes you could add an alias for the original and rename it.