nophead/NopSCADlib #295: IEC panel connector holes do not closely match IEC connector shape

cvpeck, 2024-12-26:
Hi,
I'm trying to use the IEC connectors and their associated hole templates for 3D printing of boxes. The holes for the cutouts do not exactly match the shape of the IEC connector and are a more generic hole. Whilst this does work and is better for manually cut holes, for 3D printing a closer match would be more beneficial.

[image]

cvpeck, 2024-12-26:
another option is that I am doing something incorrectly!

nophead, 2024-12-26:
I think I jut followed the datasheet:

[image]

But I agree it could be the same shape as the body to make it obvious which way round it fits.