nophead/NopSCADlib #299: DEPRECATED: Variable names starting with digits ("2a") will be removed in future releases

jeroenrnl, 2025-03-23:
When running OpenSCAD from git, I get the following warning:
```
DEPRECATED: Variable names starting with digits ("2a") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 167
DEPRECATED: Variable names starting with digits ("2c") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 168
DEPRECATED: Variable names starting with digits ("2a") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 169
DEPRECATED: Variable names starting with digits ("2c") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 169
DEPRECATED: Variable names starting with digits ("2c") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 172
DEPRECATED: Variable names starting with digits ("2a") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 172
DEPRECATED: Variable names starting with digits ("2a") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 173
DEPRECATED: Variable names starting with digits ("2c") will be removed in future releases. in file ../../NopSCADlib/utils/maths.scad, line 173
```

nophead, 2025-03-23:
Yes I won't be using the new OpenSCAD because I have relied on part numbers starting with digits for many years and projects.

nophead, 2025-03-27:
Merged at ad963afb8d636f76ef19dd96c182ab268602cc7d

closed 2025-03-27 by nophead