nophead/NopSCADlib #301: Camera position for 3D printed parts

pcremades, 2025-03-25:
Hi,
Could someone tell me how to set the camera position for 3D printed parts renders that appear in the parts list in the assemblies?

Thank you!

nophead, 2025-03-26:
Normally modules that make STL files orient them as they would be printed and that is how they appear in the pictures.

Although the comments say pose() should work for STLs it doesn't because render.py doesn't set $pose on the command line. I must have intended to implement it but forgot and never needed it yet.

I will fix it soon, perhaps tomorrow.

nophead, 2025-04-01:
Sorry for the delay, it should now work. 3715585002ba56681103103faddd043b1553d095

pcremades, 2025-04-03:
Hi Chris,
Thank you for including this feature. However, I still can't get the PNG for printed parts in the position I want. This is the code I'm using:

```scad
module base_grilla_stl() pose(a=[95, 0, 25], t=[9.75, 26, 23]){
  stl("base_grilla")
    union(){
      ...
    }
}
```

And that module is called in an assembly module like this:
```scad
module optics_assembly(){
  assembly("optics"){
     ...
     render() base_grilla_stl();
     ...
    }
}
```
Is that how you are supposed to use pose() to set the camera position for the PNG of the STLs?

nophead, 2025-04-03:
I did have to force it to be re-rendered of course as changes to the scripts don't trigger a recompilation.

pcremades, 2025-04-03:
I can't make it work. Looking through the code, I see that [at line 96 of render.py](https://github.com/nophead/NopSCADlib/blob/6a58d37f7702ac83241735ac21da509ec1ba6619/scripts/render.py#L96), cam is set to a fix value. I don't see how render.py would take camera information from pose. Am I missing something?

nophead, 2025-04-03:
The camera doesn't move. `pose()` moves the STL or assembly relative to the world, so with a fixed camera position it appears differently. In my example it gets rotated around the z-axis, but it can be translated as well.

pcremades, 2025-04-03:
Oh, I see. But when render.py creates the temp scad file for rendering, the code looks like this:

```scad
color([0, 0.572549, 0]) import("hardware/stls/base_grilla.stl");
```

shouldn't be pose() included in that line?

nophead, 2025-04-04:
Yes you are correct. I don't know how I manged to convince myself it was working It will be more difficult to implement.

nophead, 2025-04-09:
I have added a pose_stl() module to pose STLs because the default camera settings are not the same as those for assembly views and it works by changing the camera setting rather than moving the STL. Give it a try and see if it does what you want.

pcremades, 2025-04-10:
Hi Chris,
I confirm it works! Nice job. Thank you, this library/tool is fantastic. I'm closing this issue.

closed 2025-04-10 by pcremades