nophead/NopSCADlib #303: Arguments for pin headers

pcremades, 2025-04-10:
I'm trying to add a new PCB and I need to place a right angle pin header. However, [this line](https://github.com/nophead/NopSCADlib/blob/5c4fe8bfb96b54790309649d1fcfd191a8134617/vitamins/pcb.scad#L1120) seems to override parameters 3 and 4 in the call to pin_header(). Would it be ok to change that line to:

```scad
if(show(comp, "2p54header")) let($show_plugs = show_plugs && param(9, true))
                                        pin_header(2p54header, comp[4], comp[5], comp[6], comp[7], cutouts, colour = param(7, undef));
```
I've tried and it seems to work.

nophead, 2025-04-11:
param() doesn't override parameters it just provides a default if none is provided. So I don't think it needs changing. The order is not the same though, slots 7 and 8 are swapped.

pcremades, 2025-04-11:
Thank you Chris, that was the problem.

closed 2025-04-11 by pcremades