nophead/NopSCADlib #307: [Deprecation] Variable names starting with digits

pcremades, 2025-04-30:
Hi all,
I'm trying version 2025.04.27 of OpenSCAD and I receive tons of warning related to variable names starting with digits in NopSCADlib. This will become a big issue.

nophead, 2025-04-30:
Are you using the latest version? I merged a PR to fix the variables names in maths.scad. Are you seeing some from somewhere else?

pcremades, 2025-04-30:
Yes, many vitamins have names starting with digits: [2p54header](https://github.com/nophead/NopSCADlib/blob/99ff25684756ccf1ec86fd8538a03172e6ce62e9/vitamins/pin_headers.scad#L27), [6p4mm_disc](https://github.com/nophead/NopSCADlib/blob/99ff25684756ccf1ec86fd8538a03172e6ce62e9/vitamins/radials.scad#L42), just to name a few.

nophead, 2025-04-30:
Yes but I thought the warning was just for those that looked like hex constants.

pcremades, 2025-04-30:
Nop, it seems variable names in general can't begin with a digit. It will take a huge amount of work to change every single variable name in the library. I can contribute as I'm actively using NopSCADlib in several projects.
Can you figure any other way of doing this?

nophead, 2025-04-30:
Well I will just stop updating OpenSCAD rather than changing all my projects. It hasn't really been compatible with NopSCADlib since the semantics of $variable were changed just after the last release.

tpimh, 2025-05-08:
In order to just suppress the warning, we can simply prefix all variables starting with a number with an underscore. If this is fine, I can prepare a pull request with this change.

nophead, 2025-05-08:
Yes but that will break all my projects.

tpimh, 2025-05-08:
Yeah, this sucks. It's possible to fix the existing projects by running the same replacement command, or a special shim can be made that will be just full of statements like `2p54joiner = _2p54joiner`, and it would need to be included in every old project (but then again it will generate all the same warnings).

Anyway, this is not such a major concern, because it's just a warning for now, and it's going to stay this way in the next OpenSCAD release. However the release after the next release would not tolerate variable names starting with digits. Given how often the releases happen, we have years to figure out a good solution.

nophead, 2025-05-08:
The last release is the only one that works properly for me since they broke $ variables soon afterwards. I might get around to making my own fork to restore that and remove this warning. It is crazy to change semantics after a decade of people using it.

pcremades, 2025-05-08:
What if we write a Python script to automatically rename all the variables in the library? It could also be used to rename variables in any project.

P.D.: what is the problem with $ variables semantic? I'm just curious.

nophead, 2025-05-08:
One used to be able to define a $variable at the top of a file and then use it. Now it would be ignored if the file is used by another file but not if it is opened in the GUI. The change was under the disguise of refactoring but when I complained it was argued it was more correct.

To bodge around I include global_defs.scad when making STLs but that just stops the undefined variable errors. So you can't easily change say $extrusion_width for part of a project unless you do make_all with the last release. Otherwise you would need to define it in all the functions and modules of the used file. 

I can replace all the variables in my projects but I don't want to. I prefer correct part numbers.